pimoroni/pimoroni-pico-rp2350 #2: some uf2 aren't working for me

mdxtinkernick, 2024-08-10:
I have 2 tiny 2350 and 2 pico plus 2 boards

The tiny2350 uf2 is not working for me on both of the tiny boards (no board in fs mode showing in system, no serial port to connect to). pressed reset button, unplugged and plugged back in

pico_plus2_rp2350 not working for me on both of the pico plus 2 boards (no board in fs mode no serial port to connect to) pressed reset button, unplugged and plugged back in

pico_plus2_rp2350_psram working on both pico plus 2 boards ( shows up as board in fs mode, serial port shows up, connect to it and get responsive repl) sometime need to press reset to get it into that state

Gadgetoid, 2024-08-10:
Curses, I think this might be related to the PSRAM code 😫 specifically the flash write changes.

Will look into it ASAP!

Gadgetoid, 2024-08-10:
Okay, I spotted the bug.

Pretty sure I also spotted an oversight in the flash write changes, but that was not the bug.

The bug was that I am a fool who needs more sleep 🫠

mdxtinkernick, 2024-08-10:
yes, all working now thanks

closed 2024-08-10 by mdxtinkernick