pimoroni/pimoroni-pico-rp2350 #4: PSRAM configuration

Macrosii, 2024-08-16:
Not sure is it best to ask this question here or on the Pinoromi forums!  For this instance will try on both!

So just loaded the Micropython with the PSRAM image.  Thanks you very much.
Could you explain / clarify exactly whats happening.

I see that there is now around 7.8MB of free memory (around 199K used of the PSRAM 8MB ?)
For the image without PSRAM I seen around 658K free, so around 62K of the 520K SRAM.

I am concluding that everything is being run from the PSRAM  and the FLASH is not used after startup ?
If this is the case can you overclock with the limited speed of the PSRAM ?
Can you still access the onboard 520K of SRAM ?
Could you run the Python interperator in SRAM and user code in PSRAM?

Many thanks and regards

Gadgetoid, 2024-08-16:
This is really a MicroPython question, and there's a shortage of right answers because the code to add PSRAM support is not yet merged or finalised (see: https://github.com/micropython/micropython/pull/15620). Take these with a pinch of salt!

The new Pico Plus 2 image should have something like 8.2MiB free, since both the SRAM and PSRAM are combined into the same heap, and a quantity of that RAM (the calculation for which I can never seem to get right) is used by MicroPython to keep track of which RAM is in-use, for garbage collection.

MicroPython itself runs from Flash, with the notable exceptions of functions which are marked "not in flash" in the C code, these are selected to be loaded into memory for performance (or other) reasons.

User code does, indeed, run from RAM. MicroPython loads it, converts it to bytecode, and stores it in the heap. MicroPython only really has the concept of one heap (split or not) so it's not really meaningful to split user code/data between the two.

You can overclock, in theory, and the PSRAM timings should be calculated and updated accordingly. You will probably see diminishing returns on performance.

The onboard SRAM and PSRAM are merged in the latest builds, they are treated as one contiguous heap in MicroPython, though presumably (and I don't know the finer details) it avoids allocating anything that crosses from SRAM to PSRAM.

You should *possibly* load the whole MicroPython interpreter into SRAM with data/user code in PSRAM, but I suspect performance gains would be minimal.

You can also, in theory, turn off MicroPython's handling of PSRAM and bring your own code to use it how you like. If it's enabled but not added to MicroPython's heap then you can use it as a data cache, maybe mount it as a temporary filesystem, or whatever else you like. Such shenanigans are probably a bit advanced for a standard MicroPython build though.

Macrosii, 2024-08-16:
Many thanks for the very swift and comprehensive answer !

closed 2024-08-16 by Macrosii