pimoroni/pimoroni-pico-rp2350 #6: Pico Plus 2: Internal Temperature

Macrosii, 2024-09-09:
Hi
I think the Pico Plus 2 uses the QFN-80 chip.  For this chip the ADC is on the ninth ADC channel (AINSEL=8).
Micropython reports _"ValueError: Pin doesn't have ADC capabilities", for code _sensor_temp = machine.ADC(8)_

Regards

Gadgetoid, 2024-09-09:
Tested on my Pico Plus 2 and seems fine"

```python
>>> from machine import ADC
>>> ADC.CORE_TEMP
8
>>> ADC(ADC.CORE_TEMP)
<ADC channel=8>
```

Do you have a recent build of MicroPython? It's still a work in progress and things are changing quickly: https://github.com/micropython/micropython/pull/15619

Macrosii, 2024-09-09:
Hi Gagetoid

I am using the "pico_plus2_rp2350_psram" image found here so that I have PSRAM and the Pimoroni libs!
Regards

Macrosii, 2024-09-09:
OH, ok missed that _updated 6 mins ago_

Gadgetoid, 2024-09-09:
To be fair I just checked the releases after replying here (I've been using various CI builds from whatever bug or feature I'm working on) and realised there hadn't been one for three weeks 😬 , so I tagged one just now 😆 

For anyone else who might end up here: https://github.com/pimoroni/pimoroni-pico-rp2350/releases/tag/v0.0.6

Macrosii, 2024-09-09:
Yep that did it.  Many , many thanks

closed 2024-09-11 by Gadgetoid