pimoroni/pimoroni-pico-rp2350 #14: interation with SMS?

heavyimage, 2024-10-03:
Hello,

Is it possible to use the new wireless firmware + the clipper LTE module to not only get a pico online but also receive / send SMS?

Gadgetoid, 2024-10-03:
I'm pretty sure that it is, though I'm not sure if there are any relevant limitations with the SIM cards.

The downside is that it would currently involve raw AT commands and I believe you have to set up an additional command channel so that normal commands can still work while in PPP mode. I haven't explored this yet.

https://www.ktron.in/wp-content/uploads/2023/03/A76XX-Series_AT_Command_Manual_V1.06-4.pdf?srsltid=AfmBOorE2r1TGAW2YQfJeWRJXpDzKe2FOL6v9YmGLx-BOVJNm8SRemg5

heavyimage, 2024-10-03:
Thanks!  I've been playing with this for a bit and using some resources (documenting here if anyone wants to try!)

* <https://andino.systems/andino-4g-modem/sms>
* <https://github.com/pimoroni/pimoroni-pico-rp2350/commit/e5515981ea99bb0a3a349484970d6427ab8c496a#diff-2cd72d8dd88968bf8e3b41b8dde63c464074cfa5de3bcab5aa4bea66bab1da99R18>
* <https://www.smssolutions.net/tutorials/gsm/sendsmsat/?
* <https://stackoverflow.com/questions/2161197/how-to-send-receive-sms-using-at-commands>
* <http://simcomturkiye.com/pdf/4G/A7600C1/A7600_Series_AT_Command_Manual_V1.02.pdf>

...but I can't seem to get the anything but a timeout in response to the `AT+CMGS` command that starts the creation of a text message.  I think the model is supposed to reply with ">" indicating it's waiting for a message?  

Here's some very rough code / lines I've been testing with but I can't seem to get past that.  Again, for posterity: 

```python
# switch to text mode?
print("switching to text mode")
#con._send_at_command("AT+CMGF=1", 1, 10)
# read text mode values?
#con._send_at_command(b"AT+CSMP?", 1, 10)
#AT+CSMP=17,167,0,16
#print("Sending message")
#con._send_at_command('AT+CMGS="+4407808861922"', 1, 3)

recipient = "4407808861922"
message = "Hello, World!"
#con._send_at_command(b'ATZ',1 , 10)
con._send_at_command(b'AT+CMGF=1',1)
con._send_at_command(b'AT+CMGS="' + recipient.encode() + b'"',1, 20)
con._send_at_command(message.encode() + b"")
con._send_at_command(bytes([26]))
```

heavyimage, 2024-10-03:
Thanks!  I've been playing with this for a bit and using some resources (documenting here if anyone wants to try!)

* <https://andino.systems/andino-4g-modem/sms>
* <https://github.com/pimoroni/pimoroni-pico-rp2350/commit/e5515981ea99bb0a3a349484970d6427ab8c496a#diff-2cd72d8dd88968bf8e3b41b8dde63c464074cfa5de3bcab5aa4bea66bab1da99R18>
* <https://www.smssolutions.net/tutorials/gsm/sendsmsat/>
* <https://stackoverflow.com/questions/2161197/how-to-send-receive-sms-using-at-commands>
* <http://simcomturkiye.com/pdf/4G/A7600C1/A7600_Series_AT_Command_Manual_V1.02.pdf>

...but I can't seem to get the anything but a timeout in response to the `AT+CMGS` command that starts the creation of a text message.  I think the modem is supposed to reply with ">" indicating it's waiting for a message? 

Maybe is a limitation of the 'data only' SIM that came with the clipper?

Here's some very rough code / lines I've been testing with but I can't seem to get past that.  Again, for posterity: 

```python
# switch to text mode?
print("switching to text mode")
#con._send_at_command("AT+CMGF=1", 1, 10)
# read text mode values?
#con._send_at_command(b"AT+CSMP?", 1, 10)
#AT+CSMP=17,167,0,16
#print("Sending message")
#con._send_at_command('AT+CMGS="[REDACTED]"', 1, 3)

recipient = "[REDACTED]"
message = "Hello, World!"
#con._send_at_command(b'ATZ',1 , 10)
con._send_at_command(b'AT+CMGF=1',1)
con._send_at_command(b'AT+CMGS="' + recipient.encode() + b'"',1, 20)
con._send_at_command(message.encode() + b"")
con._send_at_command(bytes([26]))
```

Gadgetoid, 2024-10-04:
I think the problem is that `_send_at_command` will *always* time out if nothing is returned, but no return value is given until after the message data has been sent.

It also needs a small delay (it would seem) between CMGS and the actual message body, so we can't send it in one transaction.

Try:

```
con._uart.write(b"AT+CMGS=\"[REDACTED]\"\r")
con._uart.write(b"Hello World\x1a\r")
```

Gadgetoid, 2024-10-04:
Or, to slap that into a function (which I'll add to the lte module I think):

```python
def send_text_message(recipient, body, timeout=5.0):
    timeout *= 1000

    con._uart.write("AT+CMGS=\"")
    con._uart.write(recipient.encode("ascii"))
    con._uart.write(b"\"\r")
    time.sleep(0.5)
    con._uart.write(body.encode("ascii"))
    con._uart.write(b"\x1a\r")
    con._uart.flush()

    status, data = con._read_result(1, timeout=timeout)

    if status == "TIMEOUT":
        raise CellularError(f"cellular module timed out for SMS command")

    if status not in ["OK"]:
        raise CellularError(f"non 'OK' result for SMS command")
```

heavyimage, 2024-10-04:
Hrm, this still isn't working for me :-/

* Are you using the SIM that came with the clipper module?
* What is the syntax of the phone number you're using?  with +?  +44?  Just the number?
* I'd love to submit a PR once I get this working that also includes the ability to read messages -- how are you testing this?  Building a whole .uf2 and deploying it?

Gadgetoid, 2024-10-04:
* I am using a Lebara sim, unsure about our data ones (could be a limitation, I'll ask!)
* Both work!
* I am just sticking Python code into Thonny and testing against that

- to continue, I have a local copy of the lte.py, saved as `tlte.py` (name is arbitary) and run:

```
import gc
import tlte as lte
import time
import requests
con = lte.LTE("APN")
```

Then just hack around on the repl until something works (in Thonny, at least, the context of an executed Python script is kept and you can use those loaded functions/setup connections etc in the repl).

heavyimage, 2024-10-04:
okay thanks!  My setup is pretty similar.   I'll keep you posted :-)

And thanks for much for the help on this -- happy I was able to nerd snipe you :-)

Gadgetoid, 2024-10-04:
It is, indeed, a limitation of the sims we sell- they're data-only!

heavyimage, 2024-10-04:
AHA!   Okay.  Cool.  I think I've got another one somewhere for exactly this kind of emergency :eyes:

Gadgetoid, 2024-10-04:
Sounds like a job for an SMS -> MQTT gateway. Something like https://www.instructables.com/Two-way-MQTT-SMS-Bridge-Linkit-One/

Would be an interesting project for the LTE Pi HAT 😆

closed 2026-08-27 by Gadgetoid