pimoroni/pimoroni-pico-rp2350 #15: I need a header file for the Pico Plus 2 W for the Pico SDK

znmeb, 2024-10-15:
I don't know if this is the right repo for this issue but it's the only RP2350-specific one I could find. See also issue https://github.com/raspberrypi/pico-sdk/issues/1985.

As you probably know, the Pico SDK has a directory with header files for all the supported boards: <https://github.com/raspberrypi/pico-sdk/tree/master/src/boards/include/boards> The Pimoroni Pico Plus 2 is there but the Plus 2 W is not. Can you supply a header file in their format for the Pimoroni Pico Plus 2? Thanks!!

Gadgetoid, 2024-10-31:
It's in #10 - 

https://github.com/pimoroni/pimoroni-pico-rp2350/blob/feature/can-haz-all-the-wireless-plz/micropython/board/PIMORONI_PICO_PLUS2/pimoroni_pico_plus2w_rp2350.h

znmeb, 2024-11-02:
Thanks! I plugged this into a script I have that rebuilds all of the examples in https://github.com/raspberrypi/pico-examples for RP2350 boards on both the RISC-V and Arm cores. It runs the RISC-V cores first and it crashed very early in the run.

I can do some troubleshooting and capture the logfiles, but before I do that, should the WiFi even work on RISC-V? The regular Pimoroni Pico Plus 2 without the WiFi builds all the relevant examples for both types of core.

Gadgetoid, 2024-11-03:
> should the WiFi even work on RISC-V?

With no official reference implementation - Pico 2 W - to compare notes with, I can't possibly say. We've got a little out ahead here and are suffering for it 😬

There are also some changes coming to the WiFi driver to better support our boards/modules and I'm pulling those in ad-hoc to make our builds so I'd expect at least some trouble.

Gadgetoid, 2026-08-27:
I'm closing this as stale, but it's reminded me that we're behind the curve on upstreaming board header files so I'm sweeping our product range and fixing that.

closed 2026-08-27 by Gadgetoid