pimoroni/pimoroni-pico-rp2350 #20: lightsleep - for info

ssotheremail, 2024-11-10:
I put this comment on the Micropython github. It is also (perhaps unsurprisingly) on 0.0.9:

For me its ok as is with standard 1.24.0 on pico 1. On a Pico 2 the 1st lightsleep does not work. if I comment out the time.sleep(0.1) in the example below, then both lightsleeps work fine. Hope this helps pin it down.

import machine
import time

led = machine.Pin("LED", machine.Pin.OUT)
led.off()

while True:
led.on()
time.sleep(0.1)
machine.lightsleep(10000)
led.off()
machine.lightsleep(10000)

ssotheremail, 2024-12-05:
As far as I can see in 0.0.11 lightsleep(2500) does not wake up at all, whereas it did in 0.0.9
BTW whilst testing this:
If I put 0.0.9 or 0.0.11 on the pico I can see data files written to the pico whilst running these versions, but not files written by Micropython's "official" Micropython 1.24.1. 
If i put back 1.24.1 then files written by 1.24.1 are back, and files written by 0.0.9 andv0.0.11 are gone. If I put back the uf2 for 0.0.x they are back. No corruption is visible but....

Gadgetoid, 2024-12-05:
> No corruption is visible but....

I guess the filesystems are far enough apart to miss each other, oof. We use a 2MB/2MB split and the officially unofficial image split is still... undecided maybe...

We're still building from unmerged, unreleased changes so all bets are off until all this chaos is merged down into mainline MicroPython. Bear with us!

ssotheremail, 2024-12-06:
Thanks for all your work, and the reply. At least my battery powered project is working pretty well with its Pico 2 and brilliant Pico Inky Pack, so its hardly a crisis, but if the sleeps/timers etc were sorted then the battery life would be a lot better than its is.....

I don't envy the people trying to fix these issues, I am struggling to understand how, for example, in mainline micropython:

for y in range(0,5):
    led.off()
    #machine.lightsleep(2500)
    time.sleep(2.5)
    led.on()
    machine.lightsleep(2500) 

with the code as is with sleep(2.5) the following lightsleep produces no or a tiny delay and the loop takes about 2.5sec, whereas if you comment the sleep and uncomment the adjacent lightsleep you get a 5 sec loop! Good grief.

Gadgetoid, 2024-12-09:
Looks like this - or some flavour of it - is reported upstream too - https://github.com/micropython/micropython/issues/16181

IIRC there were patches related to sleep as part of the Pico W patchset too,  which might be what's getting pulled in here - https://github.com/micropython/micropython/pull/16057

Gadgetoid, 2026-08-27:
Closing as stale, I think the lightsleep situation is a little better with MicroPython on RP2 now, but there are still things to be fixed (upstream!).

closed 2026-08-27 by Gadgetoid