pimoroni/pimoroni-pico-rp2350 #21: Strange behaviour of onboard led sw

TonyWitts, 2024-11-17:
from machine import Pin
led = Pin("LEDW", Pin.OUT)
led.on()

hangs when run after board reset. LED turns on but REPL does not return a prompt.

If the wifi is set up first and connected then the same code works as expected without hanging.

P.S. Why "LEDW" and not "LED"?

byzantic, 2024-11-20:
Sorry to add to your woes, but I'm also trying to toggle the LED on a `pico plus 2 wireless`  using
```
>>> from machine import Pin
>>> led = Pin('LEDW', Pin.OUT)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: unknown named pin "LEDW"
>>> 
```
This is using the 0.0.10 build. At start up it reports:
`MicroPython feature/psram-and-wifi, pico2_rp2350_wireless v0.0.10 on 2024-11-18; Raspberry Pi Pico 2 (LTE + WiFi) with RP2350`

It doesn't know about `'WL_GPIO0'` either:
```
>>> wl_gpio0 = Pin('WL_GPIO0', Pin.OUT, pin_on=32, pin_cs=33, pin_dat=35, pin_clock=34)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: unknown named pin "WL_GPIO0"
```
In the release notes there's also this statement:

> Note: the RM2 module has GPIOs 0, 1 and 2 broken out, whereas the Pico Plus 2 Wireless uses GPIO0 for the onboard LED.

Which seems to contradict the other statements. Anyway, toggling GPIO0 doesn't seem to affect the LED either.

byzantic, 2024-11-21:
Doh! Of course .. it actually _says_ that at the top of the of the release notes ..

> ⚠️ Pico Plus 2 and Pico Plus 2 W users should download [pico_plus2_rp2350-v0.0.10-pimoroni-micropython.uf2](https://github.com/pimoroni/pimoroni-pico-rp2350/releases/download/v0.0.10/pico_plus2_rp2350-v0.0.10-pimoroni-micropython.uf2) - which supports WiFi and Bluetooth via either the onboard RM2 on PP2W or a SP/CE module on PP2.

Confirmed that all works nicely with the correct firmware (and that LED is _very_ bright)!

Gadgetoid, 2024-11-21:
> Doh! Of course .. it actually _says_ that at the top of the of the release notes ..

Oh no no, I added that *after* this thread to avoid others falling into the same trap!

jblanked, 2024-11-30:
Hey,

I’m using the Pico 2 W and have tried all the releases posted here: [GitHub Release](https://github.com/pimoroni/pimoroni-pico-rp2350/releases/tag/v0.0.10). 

Unfortunately, I haven’t been able to get the LED working on any of them, even when using the recommended "LEDW" trick.

To provide more details:
- **MicroPython wouldn’t load** using: `pico_plus2_rp2350-v0.0.10-pimoroni-micropython.uf2`
- **ValueError: unknown named pin "LEDW"** using: `pico2_rp2350_wireless-v0.0.10-pimoroni-micropython.uf2`
- **ValueError: unknown named pin "LEDW"** using: `pico2_rp2350-v0.0.10-pimoroni-micropython.uf2`
- **ValueError: unknown named pin "LEDW"** using: `pico2b_rp2350-v0.0.10-pimoroni-micropython.uf2`
- **ValueError: unknown named pin "LEDW"** using: `plasma2350-v0.0.10-pimoroni-micropython.uf2`
- **ValueError: unknown named pin "LEDW"** using: `tiny2350-v0.0.10-pimoroni-micropython.uf2`

Thank you for your time and assistance,  
JBlanked

byzantic, 2024-12-01:
Hi, a few things:

1. There are new releases to support the Pico 2 W. Unfortunately, the release notice on the main repository page still announces the 0.0.10 release as `latest`, but 0.0.11 releases are [here](https://github.com/pimoroni/pimoroni-pico-rp2350/releases)
2. The version you need is `pico2_w-v0.0.11-pimoroni-micropython.uf2`
3. The pin doesn't seem to be called `LEDW` - it's named `WL_GPIO0`

Transcript follows:
```python
>>> from machine import Pin
>>> led = Pin("LEDW", Pin.OUT)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: unknown named pin "LEDW"
>>> 
>>> led = Pin("WL_GPIO0", Pin.OUT)
>>> led.value(1)
>>> led.value(0)
```
There have been so many new boards issued that everyone is struggling to catch up!

jblanked, 2024-12-01:
Awesome, thank you for your help. With that version, it worked using "LED". 

```
from machine import Pin
from time import sleep
led = Pin("LED", Pin.OUT)
led.on()
sleep(3)
led.off()
```

I was only able to get the LED to work on one of the two Pico 2Ws that I ordered though. I'm assuming the LED is defective on that board. Thanks again

byzantic, 2024-12-01:
OK, yes, I think Raspberry Pi use "LED" for both the Pico and PicoW. The "LEDW" is a workaround because the Pimoroni version has to support all of Pico Plus 2, Pico Plus 2 + RM2 and Pico Plus 2 W from the same firmware.

You can check to see whether you are setting the LED pin by reading back the pin value:
```python
>>> led = Pin("LED", Pin.OUT)
>>> led.value(1)
>>> led.value(0)
>>> led.value()
0
>>> led.value(1)
>>> led.value()
1
```

jblanked, 2024-12-02:
> OK, yes, I think Raspberry Pi use "LED" for both the Pico and PicoW. The "LEDW" is a workaround because the Pimoroni version has to support all of Pico Plus 2, Pico Plus 2 + RM2 and Pico Plus 2 W [...]

The LED is being set, so it must be this board.

```
from machine import Pin
from time import sleep

led = Pin("LED",Pin.OUT)

led.on()
print(led.value())
sleep(3)
led.off()
print(led.value())
```

```
>>> %Run -c $EDITOR_CONTENT

MPY: soft reboot
1
0
>>> 
```

anglerfish27, 2024-12-03:
I've messed around with a bunch of the u2f's here, currently on the wireless, the hardware is a pimoroni pico plus 2w.

I'm rusty and going back through the fundamentals for fun I'm doing the "Getting started w/Micropython" printed by RPI.

This is a simple example, you setup an input as a pull down and then hook it up to a button who's other connection is the 3.3V rail. You do a simple while loop and it prints "you pressed the button" if you press it, or nothing if you dont. 

Well.. its goofed up, I've tried several pins and they all seem to behave the same. Yank the power plug to start fresh the value of the pin is 0 (pull down resistor), you push the button once, it changes to 1 and prints you pushed the button, then it keeps on doing that (meaning the value is jammed at 1 and saying you pushed the button when I did not.

To prove I wasn't losing my mind since at this point its stuck printing 1/button pushed even if you reset you have physically remove power from it to reset it. I hooked up an LED to the button, works like you would expect, push button, LED lights up, dont push button LED doesnt light up. Sometimes is screwed in the code. 

Like I said I tried multiple pins and the LED (not onboard) proves the code is correct. The pico is just doing whatever it wants..

The Book's code:
`from machine import Pin
from time import sleep

button = Pin(14, Pin.IN, Pin.PULL_DOWN)

# Setting up button as an input and setting it to pull down (default = 0 unpressed) so its not floating.
print(button.value())

while True:
    if button.value() == 1:
        print("You pressed the button!")
        sleep(2)`

The output even after restarting your IDE
`>>> %Run -c $EDITOR_CONTENT '`'

MPY: soft reboot
1
You pressed the button!
You pressed the button!
You pressed the button!
You pressed the button!
You pressed the button!
You pressed the button!
You pressed the button!`

here's what it looks like if you yank the cord, notice it starts off as expected then its a runaway train..

`>>> %Run -c $EDITOR_CONTENT '`'

MPY: soft reboot
0
You pressed the button!
You pressed the button!
You pressed the button!
You pressed the button!`

Thoughts? I cant make it past the first simple program in the RPI beginner book! and I've programmed a lot in the past with Picos/MP.

byzantic, 2024-12-04:
Sounds like the excess leakage issue .. https://hackaday.com/2024/09/20/raspberry-pi-rp2350-e9-erratum-redefined-as-input-mode-leakage-current/

However, it's not related to this issue, which is about pin naming.

helgibbons, 2024-12-04:
@anglerfish27 I've replied over on #27 :)

Gadgetoid, 2026-08-27:
Closing as stale!

closed 2026-08-27 by Gadgetoid