pimoroni/pimoroni-pico-rp2350 #31: Can't build for Plasma 2350W

Korhm, 2024-12-13:
Hello,

I want to build my own uf2 file to include more frozen mpy files.

I get board definition and all other files for board `PIMORONI_PLASMA2350` from tag `v0.0.10` (the provided uf2 file works as expected)

I pulled latest micropython repo and all submodules, but build fails

``` bash
➜  rp2 git:(master) pwd
/home/korhm/dev/micropython/ports/rp2

➜  rp2 git:(master) make BOARD_DIR=/home/korhm/dev/plasma_2350_w_test1/2350/board/PIMORONI_PLASMA2350 submodules
[...]

➜  rp2 git:(master) make BOARD_DIR=/home/korhm/dev/plasma_2350_w_test1/2350/board/PIMORONI_PLASMA2350 clean
[...]

➜  rp2 git:(master) make BOARD_DIR=/home/korhm/dev/plasma_2350_w_test1/2350/board/PIMORONI_PLASMA2350 
[...]
/home/korhm/dev/micropython/ports/rp2/modmachine.c: In function 'mp_machine_lightsleep':
/home/korhm/dev/micropython/ports/rp2/modmachine.c:195:38: error: 'CYW43_PIN_WL_HOST_WAKE' undeclared (first use in this function); did you mean 'CYW43_DEFAULT_PIN_WL_HOST_WAKE'?
  195 |         gpio_set_dormant_irq_enabled(CYW43_PIN_WL_HOST_WAKE, GPIO_IRQ_LEVEL_HIGH, true);
```

If I add the missing definition in .h file `CYW43_PIN_WL_HOST_WAKE`, with the same value has `CYW43_DEFAULT_PIN_WL_HOST_WAKE`, I then have same kind of issue with `CYW43_DEFAULT_PIN_WL_REG_ON`/`CYW43_PIN_WL_REG_ON`, so I also add it.

But at the very end, I have the error :
```bash
➜  rp2 git:(master) make BOARD_DIR=/home/korhm/dev/plasma_2350_w_test1/2350/board/PIMORONI_PLASMA2350 
[...]
[ 97%] Building C object CMakeFiles/firmware.dir/home/korhm/dev/micropython/lib/tinyusb/src/class/video/video_device.c.obj
[ 97%] Building C object CMakeFiles/firmware.dir/home/korhm/dev/micropython/lib/pico-sdk/src/rp2_common/pico_fix/rp2040_usb_device_enumeration/rp2040_usb_device_enumeration.c.obj
[ 98%] Linking CXX executable firmware.elf
/usr/lib/gcc/arm-none-eabi/13.2.1/../../../arm-none-eabi/bin/ld: CMakeFiles/firmware.dir/machine_pin.c.obj: in function `machine_pin_deinit':
/home/korhm/dev/micropython/ports/rp2/machine_pin.c:133:(.text.machine_pin_deinit+0x6): undefined reference to `mp_hal_is_pin_reserved'
collect2: error: ld returned 1 exit status
make[3]: *** [CMakeFiles/firmware.dir/build.make:8943 : firmware.elf] Erreur 1
make[3] : on quitte le répertoire « /home/korhm/dev/micropython/ports/rp2/build-PIMORONI_PLASMA2350 »
make[2]: *** [CMakeFiles/Makefile2:1845 : CMakeFiles/firmware.dir/all] Erreur 2
make[2] : on quitte le répertoire « /home/korhm/dev/micropython/ports/rp2/build-PIMORONI_PLASMA2350 »
make[1]: *** [Makefile:91 : all] Erreur 2
make[1] : on quitte le répertoire « /home/korhm/dev/micropython/ports/rp2/build-PIMORONI_PLASMA2350 »
-e See https://github.com/micropython/micropython/wiki/Build-Troubleshooting
make: *** [Makefile:63 : all] Erreur 1
```

I don't know what to do at this point.

Thanks.

Gadgetoid, 2024-12-17:
Right now your best bet is to fork the repository, make the changes, push them to GitHub and it should spit you out a working build. I'm slowly working on CI tooling to make our builds much easier to reproduce locally, but right now we're building from a mess of unmerged PRs anyway!

You'll also need the wireless/bluetooth branch - https://github.com/pimoroni/pimoroni-pico-rp2350/pull/18

Gadgetoid, 2025-03-05:
I've updated our CI tooling to make offline builds easier (hopefully!)

It's currently available in https://github.com/pimoroni/pimoroni-pico-rp2350/pull/18

Building locally should be a case of (assuming Linux or Linux-like, I use macOS):

```
git clone https://github.com/pimoroni/pimoroni-pico-rp2350/
cd pimoroni-pico-rp2350
mkdir build
cd build
source ../ci/micropython.sh
ci_prepare_all
ci_cmake_configure pimoroni_plasma2350
ci_cmake_build pimoroni_plasma2350
```

You may also need to run `ci_apt_install_build_deps` or manually install ccache.

Gadgetoid, 2025-11-25:
I moved all of the Plasma builds over to here and tidied them up a while back - https://github.com/pimoroni/plasma

Same better tooling, more focussed repository!