pimoroni/pimoroni-pico-rp2350 #32: Pico Plus 2 + picographics vector font scale float value < 1 causes hard lock.

AAM1130, 2025-01-08:
board: pimoroni pico plus 2
Pimoroni micropython rp2350 build v0.11 
windows 11 + Thonny

I was running some examples with a pico display pack 2.8", and I set up a loop to sample vsys and core temp. The test program ran with no issues using 'bitmap8' font at scale 2. I tried switching to 'sans' vector font. The documentation for picographics indicates display.text scale for vector fonts is a float. I changed the scale value to 0.75 hoping to roughly match bitmap8 at scale 2. After saving the file to the pico, the display showed 1 frame blip then went black, thonny showed a disconnect + permission denied error. Several attempts to reset the board resulted in a black screen and no connection. 

I was able to recover with "flash_nuke.uf2". I couldn't find any documentation showing the range limits for the scale argument, so I tested it again at scale 1.2, and no issues. And again at 0.8 there was a hard lock. 

I'm assuming this is a bug in the vector font rendering? or a division/0 error maybe?

I can provide a copy of the original main.py if there is a problem recreating the issue. I do not have any other boards with a rp2350b to compare, but I do have a standard pico 2 if additional testing would be helpful.

Gadgetoid, 2025-01-13:
I can't seem to recreate this, could you share your main.py or - ideally - a minimal reproduction, please?

AAM1130, 2025-01-13:
I started with the Pico Display Pack 2.8 example that sets up the buttons and rgb led. I then added a couple functions to sample Vsys and core-temp.  Please excuse the code, I'm still learning...  I cut away most of the additional code to try to shorten this. Hopefully its not a hardware issue. 

changing line #77 causes the lock up.
from :      display.text(f'{sample_vsys():.03f}V', 5, 20, 235, 1)
to :          display.text(f'{sample_vsys():.03f}V', 5, 20, 235, 0.8)

strangely enough... changing the other text line to 0.8 did not cause a lock up...

```
import time
from machine import Pin, ADC
import network
from picographics import PicoGraphics, DISPLAY_PICO_DISPLAY_2, PEN_P4
from pimoroni import RGBLED

# RM2 network adapter connected over SP/CE
wlan = network.WLAN(network.STA_IF, pin_on=32, pin_out=35, pin_in=35, pin_wake=35, pin_clock=34, pin_cs=33)

# Use a 4 bit/16 colour palette to save RAM
display = PicoGraphics(display=DISPLAY_PICO_DISPLAY_2, pen_type=PEN_P4, rotate=0)

# lower the backlight to 50%
display.set_backlight(0.5)

# setup display buttons inverted, 0=pressed. 
button_a = Pin(12, Pin.IN, Pin.PULL_UP)
button_b = Pin(13, Pin.IN, Pin.PULL_UP)
button_x = Pin(14, Pin.IN, Pin.PULL_UP)
button_y = Pin(15, Pin.IN, Pin.PULL_UP)

# RGB led on display
led = RGBLED(26, 27, 28)

# colors for the display
WHITE = display.create_pen(255, 255, 255)
BLACK = display.create_pen(0, 0, 0)
CYAN = display.create_pen(0, 255, 255)
MAGENTA = display.create_pen(255, 0, 255)
YELLOW = display.create_pen(255, 255, 0)
GREEN = display.create_pen(0, 255, 0)
RED = display.create_pen(255, 0, 0)

# get display bounds for locating graphics
WIDTH, HEIGHT = display.get_bounds()

# set font
display.set_font('sans')

# function to clear the frame buffer, no need to display.update()
def clear():
    display.set_pen(BLACK)
    led.set_rgb(0, 0, 0)
    display.clear()
    

# sample and convert ADC channel CORE_TEMP for mcu temp.
# On RP2350B this is ADC channel 8
def sample_temp_mcu(input=ADC(ADC.CORE_TEMP)) -> float:
    adc_volts = input.read_u16() * 3.3 / 65355
    mcu_temp_c = 27 - ((adc_volts - 0.706) / 0.001721)
    return mcu_temp_c

# sample and convert ADC channel 3 (GPIO43) for vsys.
def sample_vsys(input=ADC(43)) -> float:
    vsys = (input.read_u16() * 3) * 3.3 / 65535
    return vsys

# set up
clear()

while __name__ == '__main__':
    
    # call the func to clear the framebuffer
    clear()
    
    # Vsys
    display.set_pen(CYAN)
    display.text(f'{sample_vsys():.03f}V', 5, 20, 235, 0.8)
    
    # core temp
    display.text(f'{sample_temp_mcu():.03f}C', 5, 50, 235, 1)
    
    # show new frame buffer
    display.update()

    # delay between frames
    time.sleep(0.1)
```

Error Message in Thonny:
"Unable to connect to COM5: could not open port 'COM5': PermissionError(13, 'Access is denied.', None, 5)

If you have serial connection to the device from another program, then disconnect it there first.

Process ended with exit code 1."

I am using the only system admin account, so I'm fairly certain there are not any permissions issues. 
I tried plugging in a Yukon right after this error code and it connected with no issues, so I'm assuming its not Thonny locking up.

Gadgetoid, 2025-01-14:
Thank you. This reproduced the crash, and I managed to golf it down to:

```python
import time
from picographics import PicoGraphics, DISPLAY_PICO_DISPLAY_2, PEN_P4

display = PicoGraphics(display=DISPLAY_PICO_DISPLAY_2, pen_type=PEN_P4, rotate=0)
display.set_font('sans')
display.set_pen(display.create_pen(255, 255, 255))

while True:
    display.text('.6', x1=1, y1=0, scale=0.8)
    print("Hello")
    display.update()
    time.sleep(1)
```

It's feeling like a bug in the pen at the moment, since switching to PEN_P8 stops the crash.

But then changing `.6` to `.1` also stops it.

Or changing `x1=1` to `x1=0` so uhhh, yeah you've found a real weird 'un here!

Gadgetoid, 2025-01-14:
> I viewed the section you linked in C++. I'll need to look up a few things to fully grasp what I'm looking at.
> My knowledge of python is only intermediate and C++ is beginner at best, but if there is anything I can do to help with this please let me know.

The long and short of it is that scaled Hershey fonts trigger a `set_pixel_span` with a length of zero which, when aligned just right, caused an overflow and an attempt to write 2GB of pixels :laughing:

closed 2025-01-15 by Gadgetoid