pimoroni/pimoroni-pico-rp2350 #35: vanilla Pi Pico 2 'no module named 'pimoroni'

achrn, 2025-02-15:
Does the 'vanilla Raspberry Pi Pico 2' ufs2 file contain the pimoroni library?
On a new pico2, I add the uf2 file and:

>>> from pimoroni import RGBLED
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: no module named 'pimoroni'
>>>

alphanumeric007, 2025-02-18:
Need more info. What specific uf2 file did you use, and what example file did you run?

achrn, 2025-02-18:
I used https://github.com/pimoroni/pimoroni-pico-rp2350/releases/download/v0.0.11/pico2-v0.0.11-pimoroni-micropython.uf2, which is described as "This gives you the Pimoroni Pico libs on a vanilla Pico 2"

I ran (in the 'Shell' pane of Thonny)

> from pimoroni import RGBLED

I got:

> MicroPython feature/psram-and-wifi, pico2 v0.0.11 on 2024-11-26; Raspberry Pi Pico2 with RP2350
> Type "help()" for more information.
> >>> from pimoroni import RGBLED
> Traceback (most recent call last):
>   File "<stdin>", line 1, in <module>
> ImportError: no module named 'pimoroni'
> >>>

alphanumeric007, 2025-02-18:
Yeah, somethings broken, I'm seeing the same thing?

helgibbons, 2025-02-19:
Getting the same! 

`pimoroni.py` lives [here](https://github.com/pimoroni/pimoroni-pico/blob/main/micropython/modules_py/pimoroni.py) if you want to grab it and save it to your Pico in the meantime.

alphanumeric007, 2025-02-19:
I could have sworn I had version 11 running on a Pico2 with a Display Pack 2?
That sort of kind of worked for me, now I get
`
>>> %Run -c $EDITOR_CONTENT

MPY: soft reboot
Traceback (most recent call last):
  File "<stdin>", line 55, in <module>
AttributeError: 'Pin' object has no attribute 'read'

I get that same error 'Pin' error with V10.
`
`
import time
import random
from picographics import PicoGraphics, DISPLAY_PICO_DISPLAY_2, PEN_P8
from pimoroni import RGBLED, Button

display = PicoGraphics(display=DISPLAY_PICO_DISPLAY_2, pen_type=PEN_P8)
display.set_backlight(1.0)

WIDTH, HEIGHT = display.get_bounds()

led = RGBLED(6, 7, 8)
led.set_rgb(0, 0, 0)

from machine import Pin

button_a = Pin(12, Pin.IN, Pin.PULL_UP)
button_b = Pin(13, Pin.IN, Pin.PULL_UP)
button_x = Pin(14, Pin.IN, Pin.PULL_UP)
button_y = Pin(15, Pin.IN, Pin.PULL_UP)

# We're creating 100 balls with their own individual colour and 1 BG colour
# for a total of 101 colours, which will all fit in the custom 256 entry palette!

class Ball:
    def __init__(self, x, y, r, dx, dy, pen):
        self.x = x
        self.y = y
        self.r = r
        self.dx = dx
        self.dy = dy
        self.pen = pen

# initialise shapes
balls = []
for i in range(0, 100):
    r = random.randint(0, 10) + 3
    balls.append(
        Ball(
            random.randint(r, r + (WIDTH - 2 * r)),
            random.randint(r, r + (HEIGHT - 2 * r)),
            r,
            (14 - r) / 2,
            (14 - r) / 2,
            display.create_pen(random.randint(0, 255), random.randint(0, 255), random.randint(0, 255)),
        )
    )

BG = display.create_pen(40, 40, 40)

while True:
    
    if button_a.read():
        led.set_rgb(255, 0, 0)
    elif button_b.read():
        led.set_rgb(0, 155, 0)
    elif button_x.read():
        led.set_rgb(0, 0, 0)
    elif button_y.read():
        led.set_rgb(0, 0, 255)
    
    display.set_pen(BG)
    display.clear()

    for ball in balls:
        ball.x += ball.dx
        ball.y += ball.dy

        xmax = WIDTH - ball.r
        xmin = ball.r
        ymax = HEIGHT - ball.r
        ymin = ball.r

        if ball.x < xmin or ball.x > xmax:
            ball.dx *= -1

        if ball.y < ymin or ball.y > ymax:
            ball.dy *= -1

        display.set_pen(ball.pen)
        display.circle(int(ball.x), int(ball.y), int(ball.r))

    display.update()
    time.sleep(0.01)

`

helgibbons, 2025-02-19:
I think this issue might be specific to the Pico 2 build - the v0.0.11 build for Plasma seems to have pimoroni.py included...

@alphanumeric007 You're importing our button class there but not using it. If you're setting your buttons up as standard MicroPython pin objects you'll need to do `if button_a.value() == 0: ` to read them (logic is reversed because pull-ups).

alphanumeric007, 2025-02-19:
```if button_a.value() == 0:``` fixed things thanks. =)

achrn, 2025-02-19:
> `pimoroni.py` lives [here](https://github.com/pimoroni/pimoroni-pico/blob/main/micropython/modules_py/pimoroni.py) if you want to grab it and save it to your Pico in the meantime.

Thanks, yes that's what I did.

helgibbons, 2025-02-28:
Should be fixed in [v0.0.12](https://github.com/pimoroni/pimoroni-pico-rp2350/releases/tag/v0.0.12) :)

closed 2025-02-28 by helgibbons