pimoroni/pimoroni-pico-rp2350 #37: Are there any plans to rebase your code on micropython 1.25.0 (when avbailable) and pico sdk 2.1.1?

ssotheremail, 2025-04-07:


Gadgetoid, 2025-04-08:
Absolutely. We're actively contributing to the *development* of MicroPython 1.25.0 to add the features we need for these build 😆 (Specifically PSRAM support for Pimoroni Pico Plus 2, which was merged today)

It might lag behind the MicroPython release by a day or so, since there's a lot of busywork involved but it'll happen.

Any feature(s) you're particularly looking for?

Gadgetoid, 2025-04-08:
Note that CI builds from https://github.com/pimoroni/pimoroni-pico-rp2350/pull/18 are pretty close to what 1.25.0 will end up being. They are built from MicroPython master with some additional patches for configurable wireless (for RM2).

ssotheremail, 2025-04-09:
Thanks, good to know. 

I know that in the global scheme of things there are far bigger things than this problem on my central heating monitor project!  However there may be other people affected so.... 

My problem is machine.lightsleep. When I put the monitor together last year I used a pico inky pack (brilliant!) and the new pico2_w (want to use mqtt later) because I had read about the new 2350 power savings possibilities. It needs to run from batteries. Every minute it records various temperatures, displays them, saves them in a file, and then sleeps. 

Unfortunately if I use time.sleep the battery life is dismal.  If I use the code from https://github.com/pimoroni/pimoroni-pico-rp2350 to get the display working, machine.lightsleep hangs my code, so I have had to stick with time.sleep. There are various fixes coming through in Micropython soon, hopefully in 1.25.0 (inc #16454 which was dependent on pico sdk 2.1.1 unfortunately, so its taking a while). I have paused some other battery powered pico projects.

It would be really useful if I could have Micropython, the display and lightsleep working.  Obviously if I had known how long it would take to get lightsleep fixed, there a number of things I could have tried:

Take out the pico2w for now and try a pico, which seems to work ok with lightsleep.

Stop using the display and use a dev version of "ordinary MP" when its fixed. (Ironically the display is really useful at present as I can display a message to say the batterys have gone flat!)

Re-write it in C.

On the subject of #18 can it use the pico ink pack and a pico2w? If so, please forgive me for my poor git knowldge, can I get a uf2 or would I need to git clone it, if so how? 

Having written this it seems a totally trivial issue but thanks for your interest and and your work on Micropython.

Gadgetoid, 2025-04-09:
AIUI 1.25.0 will not include https://github.com/micropython/micropython/pull/16454, it's milestone'd for 1.26.0. Having some people actually using and testing it could be helpful, though, so I'll look in to "cherry picking" that commit over on to one of our branches so it's testable. Manually producing a MicroPython build - especially with our built-in libraries - can be tricky!

I think Inky Pack an Pico 2W should work together, the driver should be baked in to our Pico 2 builds.

Gadgetoid, 2025-04-09:
Try the Pico 2 W build from here: https://github.com/pimoroni/pimoroni-pico-rp2350/actions/runs/14353138016?pr=18

Back up your files before flashing!

I've added the following two (potentially still quite experimental) PRs to these builds:
* https://github.com/micropython/micropython/pull/16454
* https://github.com/micropython/micropython/pull/16442

ssotheremail, 2025-04-09:
Wow, thanks for this. I have downloaded (once I thought about signing in....) the pico2 and pico2w uf2s and flashed them (thanks for the reminder on existing files) on relevant devices without the pico inky hat  display. It seems you had a good idea about further testing because its certainly made a difference. 
Previously the behavior was:
Pico2 : lightsleep works correctly unless preceded by time.sleep where it returns very quickly.
Pico2w (and PicoW) Lightsleep always returns very quickly.

Now: Pico2 and Pico2w: Both now always return very quickly.

The code I am using is a fancier version of the code in my comment 9 here but with files, timing etc. I just tried the issue 9 code without the <while> on pico2w with your uf2 and its the same (but you have to time it manually). If you cant reproduce I can send the current code later.

ssotheremail, 2025-04-10:
I had a bit of time today so I tried out the "gh pr checkout 16454" on my clone as suggested by projectgus in the comments of that pr on Micropython. I haven't done 16442 like you did. "gh pr  checkout"  appeared to work and the build finished with a uf2. I put the uf2 on a pico2 without the display. Obviously I cant be 100% certain this uf2 is entirely valid but now the test still works properly for me on the pico2 with lightsleep on it own, but now lightsleep is not shortened by a preceding time.sleep any more.  This is different to your uf2 where lightsleep always returns early.  I would say that was fixed.

On the pico2_w it always returns early, as it does on your uf2. I might try  adding16442 to my clone tomorrow to see if it fixes the pico2_w early returns on the pico2_w, without breaking pico2.

Gadgetoid, 2025-09-19:
I've done a rebase. There are some pre-release builds available here: https://github.com/pimoroni/pimoroni-pico-rp2350/pull/18