pimoroni/pimoroni-pico-rp2350 #38: No display on Pico Display 2.0" with Raspberry Pi Pico 2 W

VincentPuget, 2025-04-09:
Hello,
I’m having trouble getting the **Pimoroni Pico Display 2.0"** to work with a **Raspberry Pi Pico 2 W**.

The screen’s brightness adjusts correctly when changing the parameter of the `display.set_backlight(1)` method from 0 to 1.

The LED lights up properly using the `led.set_rgb(255, 0, 0)` method.

However, I’m unable to display any text or even a rectangle on the screen.

(the screen work's perfectly on  **Raspberry Pi Pico 1**)

Firmware: `pico2_w-v0.0.12-pimoroni-micropython.uf2`

```shell
>>> import sys
>>> sys.implementation
(name='micropython', version=(1, 25, 0, 'preview'), _machine='Raspberry Pi Pico 2 W with RP2350', _mpy=7942)
```

```python
import time
from picographics import PicoGraphics, DISPLAY_PICO_DISPLAY_2
from pimoroni import RGBLED

display = PicoGraphics(display=DISPLAY_PICO_DISPLAY_2)
led = RGBLED(6, 7, 8)

display.set_backlight(1) # OK

display.clear()

while True:
    display.clear()
    my_pen = display.create_pen(255, 0, 0)
    display.set_pen(my_pen) 
    display.rectangle(20, 20, 100, 50) # not working
    display.update()

    led.set_rgb(255, 0, 0) # OK

    time.sleep(1)
```

helgibbons, 2025-04-10:
Your code is working OK for me on a Pico 2, but that `display.clear()` at the beginning of your loop is setting the background to red on the second time through the loop, making the rectangle invisible - is that intentional?

Does [our example code](https://github.com/pimoroni/pimoroni-pico/tree/main/micropython/examples/pico_display) work as expected?

VincentPuget, 2025-04-10:
I have removed that `display.clear()` but i don't still have displayed items on screen. Just the backlight and the led are working.

No, no one of your example code works on my Pico2 W.

[image]

VincentPuget, 2025-04-10:
"Could the fact that it's a Pico 2 W (Wi-Fi) be a problem?"
(On ThePiHut website, the compatibility listed for the Pico Display 2 is "Pico", "Pico W", and "Pico 2". It does not mention the "Pico 2 W".)

I just checked the screen on my Pico 1, it's working as fine.

Gadgetoid, 2025-04-10:
Is that a Pico 2 W with headers or manual soldering?

Afaik there's no incompatibility with Pico Display 2.0 with either 2 or 2 W and I think I've had mine up and running. A dodgy connection could explain a lot!

VincentPuget, 2025-04-10:
It is a Pico 2 W with headers.

I just wrote some code to test buttons input and i had input for A and B button, not for X and Y. So, i think my board has an issue.

```python
from machine import Pin
import time

button_a = Pin(12, Pin.IN, Pin.PULL_UP)
button_b = Pin(13, Pin.IN, Pin.PULL_UP)
button_x = Pin(14, Pin.IN, Pin.PULL_UP)
button_y = Pin(15, Pin.IN, Pin.PULL_UP)

while True:
    if button_a.value() == 0:
        print('Button A pressed')
    if button_b.value() == 0:
        print('Button B pressed')
    if button_x.value() == 0:
        print('Button X pressed')
    if button_y.value() == 0:
        print('Button Y pressed')
    time.sleep(0.1)
```

Thank's for your answers.

csepulv, 2025-04-11:
I have the same problem as Vincent. But all the buttons on the Pico 2W seem to be working. I also tried this with 2 different Pico 2W, so I am not sure what I might be doing wrong

helgibbons, 2025-04-14:
Managed to track down a Pico 2 W with headers to test with and the Display Pack examples are running for fine for me, using [the most recent Pico 2 W uf2](https://github.com/pimoroni/pimoroni-pico-rp2350/releases/download/v0.0.12/pico2_w-v0.0.12-pimoroni-micropython.uf2).

Might be worth double checking that the Display Pack is fully pushed down onto the Pico's pin headers?

If that doesn't help please drop support@pimoroni.com an email with a link to this issue and some photos showing how you've got everything plugged in, and we'll be able to sort out swapping the hardware.

VincentPuget, 2025-04-14:
I ordered a new model of the Pico 2 W this morning. It should arrive within a few days or weeks. I'll reopen the issue if needed. Thanks for the responses.

closed 2025-04-14 by VincentPuget