pimoroni/pimoroni-pico-rp2350 #40: Plasma 2350 examples don't work

MichaelBell, 2025-04-15:
The Plasma 2350 (W) examples do `from plasma import plasma2040`, but that gives an import error on the latest plasma2350 release uf2.

Not quite sure what the intention is here, `plasma2040` should probably be `plasma2350` anyway, but I think it did at some stage work because the pin numbers are the same.

For now the easy fix is to remove that line and replace `plasma2040.DAT` with `15` (and, if relevant, `plasma2040.CLK` with `14`).

MichaelBell, 2025-04-15:
Ah, I seem my problem is that Plasma has moved home to https://github.com/pimoroni/plasma.

But in my defence, the Plasma 2350W product page still links to this repo!

helgibbons, 2025-04-16:
Looks like I started updating everything last week, but did not finish - thanks for the nudge!

Plasma 2350, Plasma 2350 W, Plasma 2040, Plasma Stick 2040 W and Wireless Plasma Kit shop pages should all now link to the new repo.

I need to update the Learn Guide still!

helgibbons, 2025-04-16:
Learn guides now updated :)

MichaelBell, 2025-04-16:
Awesome, thank you! I'll close this issue.

closed 2025-04-16 by MichaelBell